Rem0o/FanControl.HWInfo #1: Change implementation

Rem0o, 2021-03-16:
As of version 7.0 and up of HWInfo, shared memory is limited in its free version. 

However, it looks like using the gadget feature, the sensor data can be written to the registry, which then could be imported. 

See https://docs.rainmeter.net/tips/hwinfo/

Kyguy0, 2021-03-17:
This looks great I'm sorry I can't help but it seems like it's almost there in integrating the registry read
https://github.com/PabitraPati/Windows-Registry/blob/master/Registry_Read.py

Rem0o, 2021-03-17:
@Kyguy0 that's Python. I know how to do it, just don't have the time right now, hence the help wanted.

ahodges-git, 2021-03-20:
Thanks for your efforts here, really appreciate it.

The Temperature sources from HWInfo now appear selectable as a source but all seem to be reading 0.
[image]

Rem0o, 2021-03-20:
@ahodges-git  please create a new issue with details. Your HWINFO version, config, and so on. Do you have shared memory on?

northmatt, 2021-03-22:
For those waiting for this to be updated on the registry update, you can download the portable version and run the HWiNFO32 with SHM. HWiNFO32 has no SHM limit and can be run as a separate process from HWiNFO64. So basically, use HWiNFO32 for fancontrol only (or anything else that needs SHM) and HWiNFO64 for real info needs. Do note that HWiNFO32 isnt an ideal main option for full info needs so I'd recommend only to be used for basic things like fancontrol.

thewan056, 2022-04-26:
Since this seems to have not progressed at all. I have a suggestion. Some of us are using Rainmeter to read the HWINFO registry. Can't you just use Rainmeter as a middleman and read that info from Rainmeter?

Rem0o, 2022-04-26:
@thewan056 

That would be even more work.   It's not that complicated/long to do, it just must be done.

Rem0o, 2022-05-08:
Done: https://github.com/Rem0o/FanControl.HWInfo/pull/18

closed 2022-05-09 by Rem0o