Rem0o/FanControl.HWInfo #2: Error With Plugin and HWinfo 7.0 Pro

Dazog, 2021-03-21:
3/20/2021 10:20:31 PM: Unhandled exception in FanControl v1.0.0.0
3/20/2021 10:20:31 PM: System.ArgumentException: An item with the same key has already been added.
   at System.ThrowHelper.ThrowArgumentException(ExceptionResource resource)
   at System.Collections.Generic.Dictionary`2.Insert(TKey key, TValue value, Boolean add)
   at System.Linq.Enumerable.ToDictionary[TSource,TKey,TElement](IEnumerable`1 source, Func`2 keySelector, Func`2 elementSelector, IEqualityComparer`1 comparer)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

Dazog, 2021-03-21:
Devices appear but list temperature as 0c

Rem0o, 2021-03-21:
Can you give me more context? Is it immediately when the program starts? From the exception, it seems like you got a duplicate sensor.

Dazog, 2021-03-22:
Happens when program launches.

Checked my sensors, I don't see any duplicated.

Rem0o, 2021-03-22:
I meant 2 sensors from HWInfo with the same name.

Rem0o, 2021-03-23:
Can you attach a debugger and look for a duplicate in the Load method here https://github.com/Rem0o/FanControl.HWInfo/blob/master/HWInfoPlugin.cs ?

ahodges-git, 2021-03-24:
I'm seeing the same issues as Dazog, i get the same/similar error logged continuously on startup and then whilst it's running too.

Also reads the sensors as 0c.

Running FanControl v72 with HWInfo plugin, HWInfo64 v7 with shared memory on.

24/03/2021 3:57:27 PM: Unhandled exception in FanControl v1.0.0.0
24/03/2021 3:57:27 PM: System.ArgumentException: An item with the same key has already been added.
   at System.ThrowHelper.ThrowArgumentException(ExceptionResource resource)
   at System.Collections.Generic.Dictionary`2.Insert(TKey key, TValue value, Boolean add)
   at System.Linq.Enumerable.ToDictionary[TSource,TKey,TElement](IEnumerable`1 source, Func`2 keySelector, Func`2 elementSelector, IEqualityComparer`1 comparer)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

Rem0o, 2021-03-24:
@ahodges-git again, seems like there is some kind of duplicate sensor ( with the same name ) which conflicts with the internal mechanics which expects unique names.

Rem0o, 2021-03-24:
Implemented a more robust ID system https://github.com/Rem0o/FanControl.HWInfo/commit/6755e81297bd5f5202568d3faf73ee97b07d499e

closed 2021-04-29 by Rem0o