Rem0o/FanControl.HWInfo #3: Plugin doesn't start

declain, 2021-04-05:
This is copied from log file:

5.4.2021. 12:01:38: System.IO.FileNotFoundException: Unable to find the specified file.
   at FanControl.HWInfo.HWInfo..ctor() in C:\projects\fancontrol-hwinfo\HwInfo.cs:line 38
   at FanControl.HWInfo.HWInfoPlugin.Initialize() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 18
   at FanControl.Domain.ComputerAccessLayer.Initialize(Nullable`1 disableStorage)

Rem0o, 2021-04-05:
You don't have shared memory on?

declain, 2021-04-05:
I do.

[image]

Rem0o, 2021-04-05:
Can you attach a debugger and see the problem?

declain, 2021-04-05:
I don't know how to do that, I'm not a programmer, just a regular user.

Rem0o, 2021-04-05:
Is HWInfo running (sensors) before FanControl starts?

declain, 2021-04-05:
Thank you! I started HWInfo in sensors only mode nd it works now!

closed 2021-04-05 by Rem0o