Rem0o/FanControl.HWInfo #4: v84/v85 FanControl crash

northmatt, 2021-07-01:
Full crash on startup, the process lasts for a second before exiting out.

log:
`2021-07-01 4:17:04 PM: System.NullReferenceException: Object reference not set to an instance of an object.
   at FanControl.Domain.PluginService.<LoadPlugins>d__3.MoveNext()
   at FanControl.Domain.ComputerAccessLayer..ctor(ILogger logger, IPluginService pluginService)`

Rem0o, 2021-07-01:
What plugin do you got?

northmatt, 2021-07-02:
Tried rebuilding with the latest `FanControl.Plugins` files. Use to use the one built on Mar 24th.

Files in plugin folder:
-FanControl.HWInfo.dll
-FanControl.HWInfo.pdb
-FanControl.Plugins.dll
-FanControl.Plugins.xml

Rem0o, 2021-07-02:
Remove everything but FanControl.HWInfo.dll, pretty sure FanControl.Plugins.dll being in there cause the issue as it is already loaded.  Will add protection for that.

northmatt, 2021-07-11:
I was a bit slow in trying your suggestion in v85, but did try in v86 and it worked.

closed 2021-07-11 by northmatt