Rem0o/FanControl.HWInfo #8: Plugin does not work on version 94 of FanControl

dnlrf, 2021-10-15:
I have not tried using it on previous versions. When I open the list of sensors in the graph box, only the default sensors are available. Is there a way to enable the plugin?

Below I have attached screenshots of the plugin folder and my hwinfo settings in case I did anything wrong.

[image]
[image]

user2am, 2021-12-20:
I have the same problem. I can't get Fan Control to recognize the HWiNFO64 Plugin.
I am not sure what you mean by "sensor read mode". 

I only downloaded the four files and copied them into the Plugins Folder, as seen in the second screenshot. Do I have to edit the files in any way?
Do I have to use the files from FanControl.HWInfo as well?

Thank you for your time!

Rem0o, 2021-12-20:
@user2am 
[image]

user2am, 2021-12-20:
Okay, that is what I am using.

Do we have to use your files? I only downloaded the four files from AppVeyor, as explained in the README.md. I did not use any of the files of this repo (e.g. "FanControl.HWInfo.csproj"). Maybe I am misunderstanding the installation guide?

Rem0o, 2021-12-20:
Make sure all the plugin dlls are unlocked.   Right-click => properties.
Make sure HWInfo is running b4 FanControl.

user2am, 2021-12-20:
Unlocking the dlls worked! I would have never thought of this. Thank you so much!

closed 2021-12-20 by Rem0o