Rem0o/FanControl.HWInfo #10: Sensors in FanControl display incredibly high values (then eventually 0)

johnrogers, 2021-11-15:
[image]

The plugin will work fine for a while, then eventually the sensors will start creeping up in value, eventually reaching into the thousands, then eventually becoming 0.

This is rather problematic as I am trying to base my case fan speeds on my GPU memory junction/hotspot temperatures.

Any idea what might be causing this?

chjohans, 2021-11-15:
I was gonna report the same thing. Have been using FanContgrol with the HWiNFO plugin just to let MEM Junction temperature on my RTX 3090 contribute to controlling my case fans. I have not seen this issue until just now, and I suspect the latest update of FanCOntrol that I installed earlier today as nothing else has changed.

Here are some temps that I am seeing now, some are crazy high and of course totally wrong.

I noticed this because my case fans didn't spin up and my GPU Tjunction Temp was 100 degrees C. As you can see in the second screenshot FanCOntrol reports is to be 61 degrees, but it's actually 10 degrees. FanCOntrol also thinks GPU Hopspot Temp is 100 degrees, but it's actually 74 degrees. Everything seems messed up!

See the HWiNFO screenshot for correct temps.

The issue might be FanControl itself and not the plugin, since the plugin hasn't changed but FanControl has.
[image]

[image]
[image]
ng

chjohans, 2021-11-15:
Restarted FanControl and temps now look normal, but I don't trust it after seeing this, since nothing has really changed the crazy temps might be back at any time. The HWiNFO support is great, if only it had worked.

johnrogers, 2021-11-15:
> I was gonna report the same thing. Have been using FanContgrol with the HWiNFO plugin just to let MEM Junction temperature on my RTX 3090 contribute to controlling my case fans. I have not seen this [...]

Yep, this is exactly the same thing I am seeing. It's quite problematic, since I use my 3080 to mine ETH when I'm not gaming (why not make some cash on the side). Having my case fans tied to my memory junction temperature is very important to me. I even paid for HWInfo Pro to remove the 12-hour limitation.

Rem0o, 2021-11-15:
If all you really need is junction temp, go to sensor settings, untick NvApiWrapper, and tick the GPU from LHM.  That will change the driver implementation to a version which supports junction/hotspot.

As far as these high values,  I got no idea what's happening. Each sensor is read from a set memory address/offset. If the value drift, it means somehow the location at which these get written change or something is wrong with the mapped memory.

chjohans, 2021-11-16:
Maybe just maybe it's something in FanControl or the HWiNFO plugin that causes this, especially since this issue seemed to pop up just now?

Anyhow, I took your advice and disabled HWiNFO and NvApiWrapper, the HWiNFO plugin has given me way too much trouble anyhow. Before that FanCOntrol had no issues at all, after enabling HWiNFO I've had too many issues.

It would be great to have all the sensors HWiNFO gives me access to, but GPU Mem TJunction is the most important for me that was missing and the reason I installed the HWiNFO plugin. So thanks, appreciate this sensor being made available through libreHardwareMonitor.

Rem0o, 2021-11-16:
Not in FanControl, definitely the plugin.

johnrogers, 2021-11-17:
> If all you really need is junction temp, go to sensor settings, untick NvApiWrapper, and tick the GPU from LHM. That will change the driver implementation to a version which supports [...]

Wow, thank you! Could've saved me like $30 if I'd have known sooner. Yeah, odd that this is happening. Perhaps it's an issue with HWInfo's memory sharing, who knows. Thanks for the help.

chjohans, 2021-11-17:
I'm quite sure the issue is not with HWiNFO,  I have used that for many years and never ever had any issues like this. When I said "used it", I mean used its "shared memory" function to display data in other apps.

In fact, I had a gadget on my desktop that shows some of the same data from HWiNFO using shared memory, and at the time I saw the crazy high values in FanControl that gadget, and HWiNFO itself, showed normal values. That pretty much rules out HWiNFO as the source of the error.

Rem0o, 2021-11-17:
@chjohans it's definitely the plugin as I said. Based the implementation on some code I found online, didn't spent much time on it.

chjohans, 2021-11-18:
OK thanks @Rem0o.

Rem0o, 2022-05-08:
New implementation incoming, might fix this issue.

https://github.com/Rem0o/FanControl.HWInfo/pull/18

closed 2022-05-09 by Rem0o