Rem0o/FanControl.HWInfo #14: Unable to find the specified file

darrencoen, 2022-04-05:
Placing the 4 files into the plugins directory of fan control, the log reads:

`4/5/2022 6:48:19 AM: System.IO.FileNotFoundException: Unable to find the specified file.
   at FanControl.HWInfo.HWInfo..ctor() in C:\projects\fancontrol-hwinfo\HwInfo.cs:line 38
   at FanControl.HWInfo.HWInfoPlugin.Initialize() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 18
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()
4/5/2022 6:48:19 AM: HWInfo could not initialize.
4/5/2022 6:48:50 AM: System.IO.FileNotFoundException: Unable to find the specified file.
   at FanControl.HWInfo.HWInfo..ctor() in C:\projects\fancontrol-hwinfo\HwInfo.cs:line 38
   at FanControl.HWInfo.HWInfoPlugin.Initialize() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 18
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()
4/5/2022 6:48:50 AM: HWInfo could not initialize.`

It doesn't occur in the log without the files in /plugins/. I don't have a c:\projects\fancontrol-hwinfo\ directory so I assume that's somehow hard coded?

IamGimli, 2022-04-11:
I'm having the same issue. I believe that's a problem with how the plugin available on AppVeyor was compiled as those are the directories used before the plugin is compiled.

Rem0o, 2022-04-11:
If you looked at the stacktrace a bit, which is a debug compiled version, hence the paths, you would see it doesn't find the memory mapped file, aka the Shared memory of HWInfo, meaning it's off.

IamGimli, 2022-04-11:
Thank you for the explanation, turns out you have to restart HWInfo after enabling shared memory for it to kick in, it's working now i.e. I can see the data from HWInfo in FanControl, although strangely the sensor settings page still says there's no plugin loaded.

Rem0o, 2022-04-11:
That's odd, mine does ?
[image]

IamGimli, 2022-04-11:
Yeah, it's weird, but it works, I guess.
[image]

darrencoen, 2022-04-11:
mine also does not show

[image]

darrencoen, 2022-04-11:
[image]

Rem0o, 2022-04-11:
Exit restart => still no plugin in the plugin section?

darrencoen, 2022-04-11:
> Exit restart => still no plugin in the plugin section?

correct.

Rem0o, 2022-04-11:
no logs or anything?

darrencoen, 2022-04-11:
```
4/11/2022 3:52:27 PM: System.IO.FileNotFoundException: Unable to find the specified file.
   at FanControl.HWInfo.HWInfo..ctor() in C:\projects\fancontrol-hwinfo\HwInfo.cs:line 38
   at FanControl.HWInfo.HWInfoPlugin.Initialize() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 18
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()
4/11/2022 3:52:27 PM: HWInfo could not initialize.
4/11/2022 3:54:07 PM: System.IO.FileNotFoundException: Unable to find the specified file.
   at FanControl.HWInfo.HWInfo..ctor() in C:\projects\fancontrol-hwinfo\HwInfo.cs:line 38
   at FanControl.HWInfo.HWInfoPlugin.Initialize() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 18
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()
4/11/2022 3:54:07 PM: HWInfo could not initialize.```

Rem0o, 2022-04-11:
I'm out of ideas, no clue as to how it's not displayed if the backend provider is initialized and sensors are shown.

Rem0o, 2022-04-11:
New idea:

Tick or untick any checkbox in that sensorsetting menu, click ok, then come back. Is the plugin now visible?

darrencoen, 2022-04-11:
so, i dunno if this helps or matters but for the record:

1. i completely gutted the fan control directory of any and all files and logs except the updater (which downloads everything clean)
2. the plugins directory was also untouched containing your plugin
3. hwinfo was running 
4. loading a completely fresh fan control, again it shows no plugin but it actually creates no errors

it is only when hwinfo is not running that it produces the errors i've posted and fails to initialize.

so this ghost plugin-loaded thing is maybe a fan control problem and not any issue of yours?

darrencoen, 2022-04-11:
> New idea:
> 
> Tick or untick any checkbox in that sensorsetting menu, click ok, then come back. Is the plugin now visible?

[image]

responded "no"

[image]

sensor settings first look

[image]

unchecking all about to hit ok...

[image]

yes

[image]

errors galore

[image]

progress?

Rem0o, 2022-04-11:
Yes.   You had to make a change to the sensor settings for the plugin text to be loaded in.  I got the bug figured out.

Turranius, 2022-04-30:
> ```
> 4/11/2022 3:52:27 PM: System.IO.FileNotFoundException: Unable to find the specified file.
>    at FanControl.HWInfo.HWInfo..ctor() in C:\projects\fancontrol-hwinfo\HwInfo.cs:line 38
>    at [...]

Exactly the same problem for me. It does show up just fine but refuses to load.

Rem0o, 2022-04-30:
> Unable to find the specified file.

Shared memory is not turned on or HWInfo is not on.

Turranius, 2022-04-30:
Thanks.

It was on and shared but I figured I needed to open the sensors window in HWInfo as well. Then it stopped complaining. It popped up a few new fans with new names but no new temprature sources for us in the graph curves. Weird.

Anyway, I'm happy without memory temperature. Would have been nice to toss into a mixed graph as they get warm as heck on a 3080Ti =)

closed 2022-05-09 by Rem0o