Rem0o/FanControl.HWInfo #17: v111 fan gets locked to 46.7% from program launch jarring fan ramp to 100% at program launch

clutchmaster420, 2022-04-28:
i boot my pc wait for fancontrol to launch when it launchs it takes control of fans before the profile applies to control resulting in having my fans ramping upto 100% once profile applies fans spin down but a single control refuses to go below 46.7%(gpu pump/not a pump header) until i take manual control of it then switch it back to curve control only seems to do this when launching from task scheduler aka launch at boot (max boot delay) manual launching seems the control doesnt get stuck and goes below 46.7% but fans still ramp to 100% before profile applies didnt do this back in v110
[log.txt](https://github.com/Rem0o/FanControl.HWInfo/files/8585986/log.txt)

[image]

Rem0o, 2022-05-09:
How is this related to the plugin?

clutchmaster420, 2022-05-15:
the fan ramping up seems to have fixed itself like a day before the update to v114(was still on v110) but it is still not going lower then 40% fan speed on this fan header

also what u mean by plugin im only guessing as to what is causing this annoying behavior 

[image]

[log.txt](https://github.com/Rem0o/FanControl.HWInfo/files/8695081/log.txt)

[report.txt](https://github.com/Rem0o/FanControl.HWInfo/files/8695082/report.txt)

Rem0o, 2022-05-18:
You are here: https://github.com/Rem0o/FanControl.HWInfo
Which is the repo for the HWInfo plugin.

closed 2022-07-07 by Rem0o