Rem0o/FanControl.HWInfo #19: v113 HWInfo could not initialize.

ztmzzz, 2022-05-10:
I have opened HWInfo and it works fine in v112. But in v113 and new version of this plugin , i get this log.
`2022/5/10 12:05:57: System.Exception: HWInfo is not running or reporting to gadget is not enabled.
   在 FanControl.HWInfo.HWInfoPlugin.Initialize() 位置 C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:行号 16
   在 FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()
2022/5/10 12:05:57: HWInfo could not initialize.`

Sacred-Shadow, 2022-05-10:
Same here .... 
`10.05.2022 08:54:36: System.Exception: HWInfo is not running or reporting to gadget is not enabled.
   bei FanControl.HWInfo.HWInfoPlugin.Initialize() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:Zeile 16.
   bei FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()
10.05.2022 08:54:36: HWInfo could not initialize.`

Rem0o, 2022-05-10:
Did you read the new README? It doesn't use the shared memory feature anymore.

ztmzzz, 2022-05-11:
Sorry I was too careless

closed 2022-05-11 by ztmzzz