Rem0o/FanControl.HWInfo #20: HW Info Sensors missing decimal

Boeng01, 2022-05-10:
I just updated FanControl to 1.13 and realized that the old hwinfo plugin (shared memory) didn't work any more. So i pulled the new one here, which works over the gadget implementation. In principle it works, but all sensors are missing the decimal point.

So a sensor is shown in hwinfo as 26.1°C and reported in fancontrol as 261°C. That is true for all sensors that are reported by hwinfo.

Any help to fix this would be appreciated :)

Thanks for your work, When it's working, this is by far the best solution to control fans over sensor readings.

Boeng01, 2022-05-10:
I just found a temporary solution by limiting the decimals in hwinfo to 0. Loosing some granularity here now but to control fans its more than enough to get full °C.

Anyway, something's not right here. maybe you could look into it?

Thanks

Rem0o, 2022-05-10:
Seems like a culture issue.   Comma versus dot for decimals.

I'm using float.Parse but it seems like it doesn't work on all cultures.

Boeng01, 2022-05-11:
i thought something similar. Sadly there's no option in HWinfo to change the seperator so there's no way for me as a user to fix this.
Maybe you can wrap your head around this for a future version?

Anyway, it's working good enough with full degrees, like i said. So no hurry ;)

Thanks for replying.

permissionBRICK, 2022-05-12:
Same issue, except it only occurs sometimes. When I start or restart Fancontrol, the right value for my AIO Temp is delivered (eg 28,4 C), but after a miute or two sometimes it breaks, and now the value is 28453 or similar)
I have an english install, but am using german culture settings.

permissionBRICK, 2022-05-12:
Also this seems recent, I haven't updated this plugin since I recompiled it, but it seems like either a Fancontrol, hwinfo or windows update broke this in the last 2 weeks, it was working before

Rem0o, 2022-05-18:
@permissionBRICK I released a "1.1" version of the plugin.

make sure you got the latest HWInfo version, and use the gadget stuff instead of shared memory.