Rem0o/FanControl.HWInfo #21: HWInfo Pluggin failed to load

bczegeny, 2022-06-25:
Cannot load plugin in fancontrol

Fan Control error log v 118
HWInfo Pro with shared memory access turned on

```
2022-06-24 11:03:13 PM: System.NullReferenceException: Object reference not set to an instance of an object.
   at FanControl.HWInfo.HWInfoRegistry.UpdateValues(HWInfoPluginSensor[] sensors) in C:\projects\fancontrol-hwinfo\HWInfoRegistry.cs:line 65
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 73
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   ```

Rem0o, 2022-06-26:
The plugin doesn't use the shared memory feature anymore, it uses the "gadget" feature.

See the readme here: https://github.com/Rem0o/FanControl.HWInfo

bczegeny, 2022-06-26:
@Rem0o Gadget is also turned on with all sensors. Plugin still fails to load

bczegeny, 2022-06-26:
Re-installed HWInfo, reinstalled plugin, Rebooted. 

What I find interesting about this error, is fan control is not stored at "C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs"
Why is it trying to reference this location in the log?

Or course its going to not be able to reference this file, the file doesn't exist? 
Should this path be relative to the folder of fancontroller.exe?

New error log report

```
2022-06-26 4:30:36 PM: System.InvalidOperationException: DialogHost is already open.
   at MaterialDesignThemes.Wpf.DialogHost.<ShowInternal>d__32.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at MaterialDesignThemes.Wpf.DialogHost.<Show>d__26.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at FanControl.Domain.DialogService.<>c__DisplayClass8_0`1.<<ShowDialog>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at FanControl.Domain.ApplicationContext.<InvokeAsync>d__10.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at FanControl.Domain.DialogService.<ShowDialog>d__8`1.MoveNext()
2022-06-26 4:30:37 PM: Unhandled exception in FanControl v1.0.0.0
2022-06-26 4:30:37 PM: System.NullReferenceException: Object reference not set to an instance of an object.
   at FanControl.HWInfo.HWInfoRegistry.UpdateValues(HWInfoPluginSensor[] sensors) in C:\projects\fancontrol-hwinfo\HWInfoRegistry.cs:line 65
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 73
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
2022-06-26 4:30:38 PM: Unhandled exception in FanControl v1.0.0.0
2022-06-26 4:30:38 PM: System.NullReferenceException: Object reference not set to an instance of an object.
   at FanControl.HWInfo.HWInfoRegistry.UpdateValues(HWInfoPluginSensor[] sensors) in C:\projects\fancontrol-hwinfo\HWInfoRegistry.cs:line 65
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 73
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   ```

bczegeny, 2022-06-27:
deleted fan controller, went through fan setup again. 
Saved config after it found my fans
Closed Fan Controller
Downloaded HWInfo plugin
Extracted it, placed it in the plugins folder of Fan controller.
Made sure HWInfo was running with sensors and gadgets turn on
Started Fan controller. 

Error cannot load HWInfo Plugin check error log

Error log says this:
```
2022-06-26 9:51:24 PM: System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at FanControl.HWInfo.HWInfoRegistry.GetKind(RegistryKey key, Int32 index) in C:\projects\fancontrol-hwinfo\HWInfoRegistry.cs:line 88
   at FanControl.HWInfo.HWInfoRegistry.GetSensors() in C:\projects\fancontrol-hwinfo\HWInfoRegistry.cs:line 45
   at FanControl.HWInfo.HWInfoPlugin.Load(IPluginSensorsContainer container) in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 43
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()
2022-06-26 9:51:24 PM: HWInfo could not initialize. See the error log for more details.
```

Rem0o, 2022-06-27:
What kind of sensors did you export with HWInfo? Anything that is non temperature or RPM?

bczegeny, 2022-06-27:
Sounds like you have set a max size on the array to hold sensor's

What is that array size? Maybe I need to reduce the amount of sensors?

Rem0o, 2022-06-27:
The code is here: https://github.com/Rem0o/FanControl.HWInfo/blob/master/HWInfoRegistry.cs#L89

Seems like a string exported by HWInfo can have no unit, hence no space, and thus the [1] fails.

bczegeny, 2022-06-29:
Hmm so some sensory I have has no unit and that's causing the issue, I'll double-check if there is any

bczegeny, 2022-06-29:
This is my gadget sensors not sure what one would cause an issue:

```
[Variables]
Index=0: Virtual Memory Committed -  9,734 MB
1: Virtual Memory Available -  27,811 MB
2: Virtual Memory Load -  25.9 %
3: Physical Memory Used -  7,000 MB
4: Physical Memory Available -  25,681 MB
5: Physical Memory Load -  21.4 %
6: Page File Usage -  0.0 %
84: Core VIDs -  1.363 V
7: Core 0 VID -  1.363 V
8: Core 1 VID -  1.363 V
9: Core 2 VID -  1.363 V
10: Core 3 VID -  1.363 V
11: Core 4 VID -  1.363 V
12: Core 5 VID -  1.363 V
13: Core 6 VID -  1.363 V
14: Core 7 VID -  1.363 V
85: Core Clocks -  3,898.2 MHz
15: Core 0 Clock (perf #6/7) -  3,780.1 MHz
16: Core 1 Clock (perf #4/5) -  3,780.1 MHz
17: Core 2 Clock (perf #3/4) -  3,780.1 MHz
18: Core 3 Clock (perf #1/2) -  4,725.1 MHz
19: Core 4 Clock (perf #7/8) -  3,780.1 MHz
20: Core 5 Clock (perf #2/3) -  3,780.1 MHz
21: Core 6 Clock (perf #5/6) -  3,780.1 MHz
22: Core 7 Clock (perf #1/1) -  3,780.1 MHz
23: Bus Clock -  100.0 MHz
86: Core Effective Clocks -  277.5 MHz
24: Core 0 T0 Effective Clock -  124.0 MHz
25: Core 0 T1 Effective Clock -  16.1 MHz
26: Core 1 T0 Effective Clock -  113.8 MHz
27: Core 1 T1 Effective Clock -  20.7 MHz
28: Core 2 T0 Effective Clock -  247.7 MHz
29: Core 2 T1 Effective Clock -  20.9 MHz
30: Core 3 T0 Effective Clock -  861.2 MHz
31: Core 3 T1 Effective Clock -  524.3 MHz
32: Core 4 T0 Effective Clock -  52.4 MHz
33: Core 4 T1 Effective Clock -  20.5 MHz
34: Core 5 T0 Effective Clock -  537.0 MHz
35: Core 5 T1 Effective Clock -  20.4 MHz
36: Core 6 T0 Effective Clock -  79.8 MHz
37: Core 6 T1 Effective Clock -  16.8 MHz
38: Core 7 T0 Effective Clock -  1,402.5 MHz
39: Core 7 T1 Effective Clock -  381.8 MHz
40: Average Effective Clock -  277.5 MHz
87: Core Usage -  5.2 %
41: Core 0 T0 Usage -  5.3 %
42: Core 0 T1 Usage -  0.7 %
43: Core 1 T0 Usage -  2.3 %
44: Core 1 T1 Usage -  0.7 %
45: Core 2 T0 Usage -  2.3 %
46: Core 2 T1 Usage -  0.7 %
47: Core 3 T0 Usage -  16.1 %
48: Core 3 T1 Usage -  8.4 %
49: Core 4 T0 Usage -  2.3 %
50: Core 4 T1 Usage -  0.7 %
51: Core 5 T0 Usage -  10.0 %
52: Core 5 T1 Usage -  0.7 %
53: Core 6 T0 Usage -  1.5 %
54: Core 6 T1 Usage -  0.7 %
55: Core 7 T0 Usage -  24.6 %
56: Core 7 T1 Usage -  6.9 %
57: Max CPU/Thread Usage -  24.6 %
58: Total CPU Usage -  5.2 %
89: Core Utility -  7.0 %
59: Core 0 T0 Utility -  2.9 %
60: Core 0 T1 Utility -  0.3 %
[... 931 lines cut ...]
370: Total Errors -  0 
87: Core Usage -  3.4 %
41: Core 0 T0 Usage -  4.6 %
42: Core 0 T1 Usage -  0.0 %
43: Core 1 T0 Usage -  1.5 %
44: Core 1 T1 Usage -  0.0 %
45: Core 2 T0 Usage -  0.0 %
46: Core 2 T1 Usage -  0.0 %
47: Core 3 T0 Usage -  25.3 %
48: Core 3 T1 Usage -  1.5 %
49: Core 4 T0 Usage -  0.0 %
50: Core 4 T1 Usage -  7.6 %
51: Core 5 T0 Usage -  0.7 %
52: Core 5 T1 Usage -  0.0 %
53: Core 6 T0 Usage -  2.3 %
54: Core 6 T1 Usage -  0.0 %
55: Core 7 T0 Usage -  9.2 %
56: Core 7 T1 Usage -  2.3 %
58: Total CPU Usage -  3.4 %
205: CPU -  1,005 RPM
207: Chipset -  60.8 øC
208: VDD Voltage -  1.131 V
209: NB Voltage -  1.013 V
211: NB Current -  11.294 A
```

bczegeny, 2022-06-29:
Is it just me or are you not returning a value when the temperature is in Celcius?

I mainly know javascript so maybe I'm not reading this core right
https://github.com/Rem0o/FanControl.HWInfo/blob/5bcbcfb05f731c73f06c7c73ac17d95e391ab28c/HWInfoRegistry.cs#L86
Should this function not return ` return HwInfoSensorType.Temperature;` in the ` case "°C":`
```
private static HwInfoSensorType GetSensorType(RegistryKey key, int index)
        {
            var value = (string)key.GetValue(VALUE_REGISTRY_NAME + index);
            var unit = value.Trim().Split(' ').Skip(1).FirstOrDefault() ?? string.Empty;

            switch (unit.ToUpperInvariant())
            {
                case "°C":
                // maybe should not support F since no conversion is available
                case "°F":
                    return HwInfoSensorType.Temperature;
                case "RPM":
                    return HwInfoSensorType.RPM;
                default:
                    return HwInfoSensorType.NotSupported;
            }
        }
        ```

Rem0o, 2022-06-29:
Use latest version from AppVeyor, and see if it fixes it.

bczegeny, 2022-06-29:
FIXED! Finally loading correctly!

closed 2022-06-29 by bczegeny