Rem0o/FanControl.HWInfo #23: HWInfo sensors were changed during operation

Fortiori, 2022-07-16:
I'm getting a recurring issue where I receive the following error in Fan Control (v120). Of note I'm reading from an ASUS Z690 EC module and HWinfo does warn that there might be issues. Refreshing sensor detection in Fan Control fixes the issue so maybe a refresh-on-error could workaround it?

`16-Jul-22 19:48:39: Unhandled exception in FanControl v1.0.0.0
16-Jul-22 19:48:39: System.Exception: HWInfo sensors were changed during operation
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 68
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)`

Rem0o, 2022-07-18:
The message tells exactly what's happening. Some sensors stopped reporting from HWInfo. Make sure you only got the sensors you need exported from HWInfo. Some users check all the sensors checkbox, but there seem to be a consistency issue with value reported in the registry

actioninja, 2022-07-24:
Having the same issue, only exporting one sensor that I need from HWInfo. Nothing is changing on my system.
```
7/23/2022 10:31:57 PM: Unhandled exception in FanControl v1.0.0.0
7/23/2022 10:31:57 PM: System.Exception: HWInfo sensors were changed during operation
   at FanControl.HWInfo.HWInfoPlugin.Update()
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   ```

Fortiori, 2022-07-24:
I gave up, couldn't find an issue in HWInfo and kept getting the same error - only reporting 2 values. Unfortunately that is the only way I could get water temp with via the ASUS EC on the mobo so I'm using other temps for now.

ArbiterGR, 2022-08-18:
I too was experiencing the same issue.  But i think i've found out the solution. In HWinfo Sensors settings, i enabled "show all fans (includding stopped or not present)" and "accept negative temperatures". I'ts been 16 hours now without the error popping up. My guess is that the hwinfo widget somehow glitches momentarely not showing the sensors and Fancontrol gets triggered by that.

Rem0o, 2022-08-18:
@ArbiterGR  oh god you got it: As soon as a fan stops (0 rpm), HWInfo doesn't push the updates in the registry.

ArbiterGR, 2022-08-18:
Thank you for the explanation, It makes so much sense now. It was driving me crazy until i stumbled upon that setting in hwinfo.

In my case, the fans do not stop. I monitor NZXT Kraken Z73's Liquid temp, fan and pump speeds. Yet somehow, hwinfo was randomly removing the sensor from the registry for just a moment, making FanCntrol to trigger the error. I suspect that this could be due to a communication error between hwinfo and the AIO's usb controller (but that's irrelevant).

ItsMeAubey, 2022-09-24:
Resolved by #29

closed 2022-09-24 by Rem0o