Rem0o/FanControl.HWInfo #26: HWInfo could not initialize or has no sensors.

Platinum24000, 2022-07-29:
I saw an issue where you said to fix it by downloading the build from Appveyor but I honestly have no idea how to do that or how that site works. Could you upload a new release so I can just download the updated FanControl.HWInfo.dll file directly?

Thanks!

Rem0o, 2022-07-30:
Latest release is up to date.

Platinum24000, 2022-07-30:
I just redownloaded it and put it into the plugins folder but its not even showing the plugin after I refresh sensors detection

Platinum24000, 2022-07-30:
Also, there is no artifacts on the Appveyor page. The release I downloaded is from May 18th.

KamiYamabushi, 2022-08-05:
Just like to chime in that FanControl no longer recognizes the HWInfo plugin either.

[image]
[image]

Rem0o, 2022-09-11:
@KamiYamabushi  you only need the FanControl.HwInfo.dll file in there.

Also, make sure it is not locked (right-click, properties)

KamiYamabushi, 2022-09-14:
> @KamiYamabushi you only need the FanControl.HwInfo.dll file in there.
> 
> Also, make sure it is not locked (right-click, properties)

Unfortunately, FanControl is still not detecting the HWInfo plugin even with nothing being locked and it be the only file in the directory.

Th4nato5, 2022-09-17:
> > @KamiYamabushi you only need the FanControl.HwInfo.dll file in there.
> > Also, make sure it is not locked (right-click, properties)
> 
> Unfortunately, FanControl is still not detecting the HWInfo plugin even with nothing being locked and it be the only file in the directory.

You might need to click unblock in properties, and then restart Fan Control. It works for me after doing this.

donnysmit, 2022-10-04:
I have the same issue, got all latest versions (HWMonitor, HWInfo Plugin & FanControl). `FanControl.HWInfo.dll` is not blocked and placed in the plugins folder.

"HWInfo could not initialize or has no sensors."

marxzui, 2022-10-30:
> Just like to chime in that FanControl no longer recognizes the HWInfo plugin either.
> 
> [image] [image]

I had this problem when I started Fan Control through a desktop shortcut. If I started Fan Control directly in the installation folder, the plugin was found then. I added a working directory to the shortcut and then the plugin was also found when starting by shortcut.

Rem0o, 2022-10-30:
^

FanControl uses relative paths. So yes it needs to have the working directory set to its own directory to find plugins and everything to work properly.

ZiOLONwe, 2023-01-04:
I have problem with this comunicat "HWInfo could not initialize or has no sensors".

testc2n14, 2024-03-06:
i'm having the same issue and i can;t figure out how to fix it