Rem0o/FanControl.HWInfo #28: Refuses to load, dependency not found?

JajajTec, 2022-09-11:
[log.txt](https://github.com/Rem0o/FanControl.HWInfo/files/9543307/log.txt)
Fan controll v127
Plugin doesent want to load
Activated HWinfo gadget reporting (and restarted it)
Did i muss something?

PS: sorry for german log file. But i dont know of a way to translate a whole log file...

Rem0o, 2022-09-11:
Make sure to unlock the dll (right-click, properties...).

JajajTec, 2022-09-11:
oke wow, dident think that windows does that. it works now. thanks alot!
i tryed it out now and saw that it wont show wattage in sensors even if you activate it in hwinfo (temperatures work tho)
is there any other way i can include wattage (cpu maybe gup later) as a sensor?

Rem0o, 2022-09-11:
FanControl has no wattage support.

JajajTec, 2022-09-11:
:(
so i wouldent even be able to do it by using that costume file feature?
btw. how would that work. how has that .txt to be formatted? just the value, that it?

Rem0o, 2022-09-11:
> how would that work

Value as a string, that's it.

closed 2022-09-12 by JajajTec