Rem0o/FanControl.HWInfo #32: HWInfo sensor value went missing from registry

zurlox, 2022-10-23:
Really liking 'Fan Control' and the plugins that are available. But seem to be having issues with the HWInfo plugin.
Plugin loads and sensors appear in the app briefly then Fan Control throws an error, all HWInfo sensor values disappear and the below log entries are recorded:

23/10/2022 6:07:05 PM: Unhandled exception in FanControl v1.0.0.0
23/10/2022 6:07:05 PM: System.Exception: HWInfo sensor value went missing from registry

I've tried reloading the sensors, replaced the .dll file from the one available on AppVeyor, however still getting the unhandled exception
Any thoughts on what I can try to fix this?
Thank you very much and keep up the awesome work!

Rem0o, 2022-10-23:
Few things to keep in mind.

1) limit the quantity of sensors you export from HWInfo to the ones you use.
2) this setting will remove 0 rpm fans if left unchecked and may cause this issue.
[image]

zurlox, 2022-10-24:
Thank you for your prompt reply.
I've made sure that 'show all fans' options is enabled, and I've reduced the sensor quantity considerably.
The error message is still displayed in FanControl when the plugin is enabled.
Is there maybe anything else I can try to resolve?
Thank you again,

Rem0o, 2022-12-01:
> HWInfo sensor value went missing from registry: HWInfo/Corsair Platinum/CPU Cooler Coolant Temperature/

Seems like at some point HWInfo stopped reporting that value for more than 10 seconds. Is that expected?