Rem0o/FanControl.HWInfo #34: HWInfo could not initialize or has no sensors

XYZliang, 2023-01-11:
I installed a adjustable fan on the memory of the chassis. I want to control the fan by monitoring the temperature of the memory, but the FanControl does not have the temperature of the memory I need, so I try to use your plug-in.
I downloaded the latest version of FanControl(V142), FanControl.hwinfo (1.2), HWiNFO64(7.34-4930),
After following the steps, FanControl can recognize the HWInfo plug-in and I have enabled it. At the same time I started HWiNFO in sensor-only mode and checked the "Enable reporting to Gadget" check box while selecting "Report Values in Gadget" for the memory temperature sensor.
But nothing seems to have changed in FanControl. When I tried "Refesh sensors detection", FanControl showed "HWInfo could not initialize or has no sensors".
[image]

Rem0o, 2023-01-11:
Any logs.txt? 

Also, take a look at your windows registry, this location:
Computer\HKEY_CURRENT_USER\Software\HWiNFO64\VSB

There should be an entry in there per sensor you export for HWInfo.

XYZliang, 2023-01-19:
我在关闭了其他插件仅开启HWInfo并删除旧的log文件后，尝试刷新传感器数据甚至是重启FanControl，在出现无法初始化的错误后，我发现FanControl并没有输出任何log文件内容。
注册表目录内包含了我所需要的条目：
[image]

Rem0o, 2023-01-19:
I suspect the "non ascii" entries ( is that chinese ? ) causing problem.

XYZliang, 2023-01-20:
> 我怀疑“非ascii”条目（是中文吗？）引起了问题。

Thank you for your reply, as you guessed, it is in Chinese. I changed the name of the entry, but the problem persists.
[image]

Rem0o, 2023-01-20:
Any Logs.txt in fancontrol's folder?

virtuoso-00, 2023-02-05:
Had the same issue. Easy fix after looking at the log:

[image]

MiyukiLoveu, 2023-02-21:
same issue
[image]
[image]
[image]

Rem0o, 2023-02-21:
@MiyukiLoveu  make sure HWInfo is open in sensor mode, with some sensors exported to gadget.

chaoyue1024, 2023-02-23:
> 

你好，你最后解决了不能识别内存温度的bug了吗，我碰到了一样的情况。日志里面没看懂，就一行HWInfo could not initialize or has no sensors.

chaoyue1024, 2023-02-23:
Had the same issue.
[image]

MiyukiLoveu, 2023-02-24:
> @MiyukiLoveu make sure HWInfo is open in sensor mode, with some sensors exported to gadget.

Thank you , it works

juzi5201314, 2023-02-28:
same issue. already exported the sensors to the gadget. The log only shows `HWInfo could not initialize or has no sensors.`

XYZliang, 2023-03-03:
> same issue

same issue and same log

mumyeong54, 2023-04-02:
Adding power[W] gives the same error.
HWInfo -> Configure Sensors -> Customize, solved in changed units.

MojoMC, 2023-04-12:
Same problem here.

Log entry: "HWInfo could not initialize or has no sensors."
HWInfo is running in sensor mode, reporting to gadget is enabled and individual sensors are ticked off to report the values to gadget.
The entries in the registry at HKEY_CURRENT_USER\Software\HWiNFO64\VSB are there, without non-ASCII characters.
Latest version of FanControl (V153), latest version of FanControl.HWInfo (V1.3). Plugins-Folder created in FanControl and FanControl.HWInfo.dll copied there. The dll is not blocked/locked.
FanControl is started from the folder, not from a desktop shortcut.

Any ideas?
I need HWInfo because LibreHardwareMonitor isn't finding any sensors at all.

pheare, 2023-04-13:
I am seeing this issue as well for every sensor I try other than CPU (Tctl/Tdie).

In hWiNFO Gadget tab if I have CPU (Tctl/Tdie) selected (Report value in Gadget), I do not get the "HWInfo could not initialize or has no sensors." error, but none of the other sensors I've selected (GPU Core Load, for example) are selectable in Fan Control.

If I remove CPU (Tctl/Tdie) in hWiNFO Gadget then I get the error - even though I still have other sensors selected.

MojoMC, 2023-05-17:
> In hWiNFO Gadget tab if I have CPU (Tctl/Tdie) selected (Report value in Gadget), I do not get the "HWInfo could not initialize or has no sensors." error, but none of the other sensors I've selected (GPU Core Load, for example) are selectable in Fan Control.

I would try this but my platform does not offer those sensors in HWInfo.

@Rem0o Any updates on this issue or suggestions how we can assist to further troubleshoot this problem?

Milkiananas, 2023-05-18:
Some comments: the unit for the sensor readings has to be °C, and if not, can be renamed it in HWiNFO.

Fooughhy, 2023-05-20:
Have we always had to enable "Report value in Gadget"? I thought I had used this plugin before, but I have never enabled any values in the "HWiNFO Gadget" tab at all.

I did that now, and got this error when I refresh sensors in Fan.Control:

> 2023-05-20 06:46:30: Unhandled exception in FanControl v157.0.0.0
2023-05-20 06:46:30: System.Exception: HWInfo sensors failed: HWInfo/GPU [#0]: NVIDIA GeForce RTX 3070: /GPU Hot Spot Temperature/°C
   vid FanControl.HWInfo.HWInfoPlugin.Update() i C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:rad 93
   vid FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   vid System.Collections.Generic.List`1.ForEach(Action`1 action)
   vid FanControl.Domain.ComputerAccessLayer.Update()
   vid FanControl.Domain.ApplicationClock.DoActions()
--- Slut på stackspårningen från föregående plats där ett undantag utlöstes ---
   vid System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   vid System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   vid System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

Also, the first response from @Rem0o :

> Any logs.txt?
> 
> Also, take a look at your windows registry, this location: Computer\HKEY_CURRENT_USER\Software\HWiNFO64\VSB
> 
> There should be an entry in there per sensor you export for HWInfo.

I do not have a "folder" called "VSB" there. I only have "Sensors" and "Summary". Is that incorrect?

abcdehc, 2023-06-30:
i have vsb and value, my problem is same with @XYZliang 
i guess its becaues the some formatting is incorrect, maybe a separator, maybe a unit of measurement.
[image]
[image]
[image]
[image]

XYZliang, 2023-07-08:
@Rem0o I recently reinstalled my system and tried to re-use the plugin.
The problem is the same as before, but this time the logs have some information about the error reported as:
"2023/7/8 22:53:31: System.Exception: HWInfo is not running or reporting to gadget is not enabled.
   At FanControl.HWInfo.HWInfoPlugin.Initialize() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 27
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()"
But I'm sure hwinfo is running and has two temperatures open. I can see the corresponding entries in the registry HKEY_CURRENT_USER\Software\HWiNFO64\VSB

XYZliang, 2023-07-08:
> 我有 VSB 和值，我的问题与我想它的原因相同，因为某些格式不正确，也许是分隔符，也许是度量单位。 [image] [image] [image] [image]

你的日志文件有内容吗？我之前使用aida64插件替代，但aida内存显示经常不稳定，还是hwinfo的检测靠谱，可是这个插件还是没法用，日志现在的报错说我没开报告功能

XYZliang, 2023-07-08:
After several flushes, you get the following log file:

2023/7/8 22:53:31: System.Exception: HWInfo is not running or reporting to gadget is not enabled.
   at FanControl.HWInfo.HWInfoPlugin.Initialize() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 27
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()
2023/7/8 23:06:24: Unhandled exception in FanControl v161.0.0.0
2023/7/8 23:06:24: System.Exception: HWInfo was closed during operation.
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 107
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location ---
   at FanControl.Domain.ApplicationClock.<>c__DisplayClass10_0.<DoActions>b__0()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
2023/7/8 23:06:39: System.Exception: HWInfo is not running or reporting to gadget is not enabled.
   at FanControl.HWInfo.HWInfoPlugin.Initialize() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 27
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()

ouyangyiluo, 2024-01-03:
Try this version https://github.com/Rem0o/FanControl.HWInfo/commit/c166d1467555ae702e0b979143c711835ef6643e

haze1986, 2025-02-04:
Hi, issue persist in v1.4, having the readings in registry but fan control aren't importing them