Rem0o/FanControl.HWInfo #35: HWiNFO data handling issue

brunoherrera, 2023-01-18:
I had an issue with liquidctl plugin, and had to go back to HWiNFO, and found out the real cause of [this issue](https://github.com/Rem0o/FanControl.Releases/issues/1452)

The issue mentioned above was that logs.txt was constantly being written, I thought it was just HWiNFO being weird about it, but at the end, I think Fan Control has the issue receiving data from it.

As long as you first enable other sensor reporting from HWiNFO Gadget, and then enable liquid temperature sensor reporting so it's not indexed as "0" it will stop logging to logs.txt, so maybe it's the way the data is being handled? I don't know, but is a solid, I now have it working perfectly, I did change it back so index is 0 and indeed starts to write logs.txt every second, confirmed Index is the issue

[With this index works perfectly](https://i.imgur.com/MbrgCRe.png)
[With this index it constantly writes to logs.txt](https://i.imgur.com/tcu30TZ.png)

Rem0o, 2023-01-18:
The interesting part is why it logs in the first place:
https://github.com/Rem0o/FanControl.HWInfo/blob/6ec922093d6e307d70dd929cfa1294944a814a4b/HWInfoPlugin.cs#L88

The app updates every second.
As you can see, it means that said sensor simply went missing in the registry during that update cycle, the value of that registry key could not be parsed to a float.  My guess is the parsing. The value must be something I'm not expecting/never seen before.  Would be practical to know what that is.

Rem0o, 2023-01-18:
Just did a commit 
https://ci.appveyor.com/project/Rem0o/fancontrol-hwinfo/builds/45979464

It will only log the failed sensors once you close the app or refresh the app.
I added more log information so we can take a look at the value it couldn't parse or if the sensor simply disappeared.

brunoherrera, 2023-01-18:
Just tried it and no luck, you will receive an email with a screen recording video in a few minutes so you can see how I have it setup and maybe there's something I didn't notice, also started logging sensors in HWiNFO and there wasn't any weird value in the csv

Rem0o, 2023-01-19:
Did you use the binary from appveyor?

brunoherrera, 2023-01-19:
> Did you use the binary from appveyor?

of course, I sent you a link to youtube with the recording, I created a copy of my Fan Control and unzipped the new contents there, 2 already existed and clicked replace so I had all the new 4 files from appveyor

Rem0o, 2023-01-19:
From the video, you didn't put the DLL in the plugin folder, you put it in the FanControls root folder. Also, from the text log we see later, the message being shown is not possible if you used the latest binary: the message was deleted from the code.

brunoherrera, 2023-01-19:
oh, also, if you noticed from the recording, it grays out every 2 seconds in hwinfo and when it does that the registry key also gets deleted, and is the only sensor (that I use) that has that issue

brunoherrera, 2023-01-20:
@Rem0o caught this, in case it's of any help.

```
1/20/2023 01:20:28: HWInfo sensor failed momentarily during operation: HWInfo/NXZT Kraken X53/X63/X73/Liquid Temperature/°C - Missing
1/20/2023 01:20:28: Unhandled exception in FanControl v1.0.0.0
1/20/2023 01:20:28: System.Exception: HWInfo sensors failed: HWInfo/NXZT Kraken X53/X63/X73/Liquid Temperature/°C
   at FanControl.HWInfo.HWInfoPlugin.Update()
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
```

Rem0o, 2023-01-20:
^ that means the sensor was undetected for more than 10 cycles straight, which at that point I would say it is probably not reliable enough to be used as a source.

brunoherrera, 2023-01-20:
at least now it isn't writing every second to log.txt with your fix :) thanks!

brunoherrera, 2023-01-25:
found out it is HWiNFO current implementation of NZXT Kraken devices they added in last version, edited the issue with a workaround until they fix it (if at all).

closed 2023-01-25 by brunoherrera