Rem0o/FanControl.HWInfo #36: NZXT Sensor via HWInfo causing exception and stopps reading temp

TheAutistCoder, 2023-02-19:
NZXT and CAM is a sh*tshow all by itself. They tried to prevent 3rd party apps from accessing controls and sensors.

However, via HWInfo and the FanControl plugin, it's possible to read e.g. the liquid temp sensor. But it doesn't last long. That's when HWInfo is able to re-establish reading the temp by (I guess) deleting and re-initializing the sensor. And at that point, FanControl stops working/updating the temp reading with the following exception:

```
19.02.2023 11:30:50: Unhandled exception in FanControl v1.0.0.0
19.02.2023 11:30:50: System.Exception: HWInfo sensors were changed during operation
   bei FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:Zeile 68.
   bei FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   bei System.Collections.Generic.List`1.ForEach(Action`1 action)
   bei FanControl.Domain.ComputerAccessLayer.Update()
   bei FanControl.Domain.ApplicationClock.DoActions()
--- Ende der Stapelüberwachung vom vorhergehenden Ort, an dem die Ausnahme ausgelöst wurde ---
   bei System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   bei System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   bei System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
```

If I click "Refresh Sensor Detection", it will work another 1 to 10 Minutes. Sometimes longer, before the next exception occurs.

It seems the sensor's path didn't change (I don't need to reconfigure the sensor in FanControl). However, it looks like HWInfo removed and re-added the sensor, which caused the exception and stopped the reading.

I guess the solution is to allow sensory to disappear and reappear after a few update cycles without causing an exception. Unfortunately, my last C# contact was many years ago. Otherwise I'd have given it a shot. But maybe you have a quick and clean solution for this?

Rem0o, 2023-02-19:
Realized you are not using the latest version of the code. Haven't made a release for a while.  Just did.  Try and use the latest version and report back.

TheAutistCoder, 2023-02-20:
Thanks, I'll try and report back / close the issue within the next 2 days.

TheAutistCoder, 2023-02-20:
Ok. The problem is not solved completely. I found this exception in the log after starting back up from standby/hybernation...

```20.02.2023 08:59:09: HWInfo sensor failed momentarily during operation: HWInfo/NXZT Kraken Z63/Z73/Flüssigkeitstemperatur/°C - Missing
HWInfo/NXZT Kraken Z63/Z73/Lüfter/RPM - Missing
HWInfo/NXZT Kraken Z63/Z73/Pumpe/RPM - Missing
20.02.2023 09:33:12: HWInfo sensor failed momentarily during operation: HWInfo/NXZT Kraken Z63/Z73/Flüssigkeitstemperatur/°C - Missing
HWInfo/NXZT Kraken Z63/Z73/Lüfter/RPM - Missing
HWInfo/NXZT Kraken Z63/Z73/Pumpe/RPM - Missing
20.02.2023 09:33:12: Unhandled exception in FanControl v1.0.0.0
20.02.2023 09:33:12: System.Exception: HWInfo sensors failed: HWInfo/NXZT Kraken Z63/Z73/Flüssigkeitstemperatur/°C, HWInfo/NXZT Kraken Z63/Z73/Lüfter/RPM, HWInfo/NXZT Kraken Z63/Z73/Pumpe/RPM
   bei FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:Zeile 93.
   bei FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   bei System.Collections.Generic.List`1.ForEach(Action`1 action)
   bei FanControl.Domain.ComputerAccessLayer.Update()
   bei FanControl.Domain.ApplicationClock.DoActions()
--- Ende der Stapelüberwachung vom vorhergehenden Ort, an dem die Ausnahme ausgelöst wurde ---
   bei System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   bei System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   bei System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)```

Also, the liquid temp now comes out as "not a number":
![grafik](https://user-images.githubusercontent.com/125797732/220120871-7832d5a3-abd1-4ebb-9219-f9dc4e37f75d.png)

After refreshing sensors, FanControl was able to read the temp and update it. I'll keep you updated about further exceptions and how it behaves after reboots / normal shutdown & boot.

TheAutistCoder, 2023-02-20:
Update: The exception re-occurs randomly during normal operation, and the temp sensor shows "NaN" until I manually refresh sensors.

Rem0o, 2023-02-20:
From your log, it means the sensor goes offline from HWInfo for more than 10 seconds, and from the plugin POV that's unreliable enough to throw an error.

TheAutistCoder, 2023-02-21:
Ok. I can think of some more or less elegant solutions for this, which would probably result in a UI/UX nightmare.

I think I will fork a version to add some debugging stuff, like....

- Introduce a separate fail counter for devices having "NZXT" and/or "Krakken" in their name.
- Addidtional logging when a previously missing sensor is back.
- When the sensors fail, I fake the temp, like: `fakeTemp = 30 + _updateFailCountNzxt`, so the fans ramp up.

Maybe that will give me an idea for a user friendly / non-coder solution. If you are interested, you can keep this issue open or I can create a pull request if I come up with something usefull.
Thanks anyway! =)

TheAutistCoder, 2023-03-16:
Hi again!

I could completely eliminate the issue by changing your code (HWInfoPlugin.cs:90) to `if (++_updateFailCount >= 30)`. Changing it to 20 didn't always succeed. It was better, but not eliminated. I only tried 20 and 30. And I didn't make a distinction for NZXT devices.

I thought about only applying 30 if the missing sensor IDs contain one with "NZXT" in the name, but NZXT may not be the only manufacturer with this problem. As far as I know they are also only licensing their hardware from another manufacturer.

Hope I could help. Maybe you can figure out something more fancy. ;)