Rem0o/FanControl.HWInfo #37: NaN with HWiNFO plugin

lgraak, 2023-03-26:
Having an issue where win using the HWiNFO plugin, just getting NaN as the data from HWiNFO. FanControl itself isn't seeing the temp sensor that's connecting to my water temp. Would like to run my radiator fans off what the water temp is, not cpu or gpu temp.

Made sure HWiNFO gadget settings are correct.

[image]

Rem0o, 2023-03-26:
HWInfo, while running, exports these sensors valyes into the registry at this address: 

Computer\HKEY_CURRENT_USER\Software\HWiNFO64\VSB

Like so:

[image]

What do you get?

lgraak, 2023-03-26:
I exported it out and saved as a txt file.

[hwinfo.txt](https://github.com/Rem0o/FanControl.HWInfo/files/11073243/hwinfo.txt)

Rem0o, 2023-03-26:
You exported basically everything? Try and only export the temperature sensors you need like t_sensor.

lgraak, 2023-03-26:
"Sensor502"="ASUS EC: ASUS "
"Label502"="T_Sensor"
"Value502"="84.2 °F"
"ValueRaw502"="84.2"
"Color502"="004080"

lgraak, 2023-04-16:
Looks like it's an issue with gpu items. Removed those items from the gadget and it seems to fix the issue

`4/15/2023 7:31:00 PM: System.Exception: HWInfo sensors failed: HWInfo/GPU [#0]: NVIDIA GeForce RTX 4080: /GPU Performance Limiters/, HWInfo/GPU [#0]: NVIDIA GeForce RTX 4080: /Performance Limit - Power/, HWInfo/GPU [#0]: NVIDIA GeForce RTX 4080: /Performance Limit - Thermal/, HWInfo/GPU [#0]: NVIDIA GeForce RTX 4080: /Performance Limit - Reliability Voltage/, HWInfo/GPU [#0]: NVIDIA GeForce RTX 4080: /Performance Limit - Max Operating Voltage/, HWInfo/GPU [#0]: NVIDIA GeForce RTX 4080: /Performance Limit - Utilization/, HWInfo/GPU [#0]: NVIDIA GeForce RTX 4080: /Performance Limit - SLI GPUBoost Sync/
   at FanControl.HWInfo.HWInfoPlugin.Update()
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at FanControl.Domain.ComputerAccessLayer.<>c.<Update>b__33_0(IBackendProvider x)
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location ---
   at FanControl.Domain.ApplicationClock.<>c__DisplayClass10_0.<DoActions>b__0()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)`

[image]

Ketoyu, 2023-07-01:
I had a similar issue when swapping out a hard drive (usually requires me to re-select sensors) but this time it got stuck in a loop when I removed the old and new hard drives (leaving just my M.2s). I could interact with most of the interface, but the actions (new configuration, refresh sensor, etc.) didn't seem to do anything.

I managed to resolve it by 1) stopping the process, 2) renaming the active configuration file, 3) starting the program again, and 4) manually "loading" the renamed configuration file. I was then able to re-select the sensors like before.

Ketoyu, 2023-07-01:
I had a similar issue when swapping out a hard drive _(usually requires me to re-select sensors)_ but this time it got stuck in a loop when I removed the old and new hard drives _(leaving just my M.2s; new hard drive wasn't compatible & old was too big a hassle too put in rn)_. I could interact with most of the interface, but the actions _(new configuration, refresh sensor, etc.)_ didn't seem to do anything.

I managed to resolve it by 1) stopping the process, 2) renaming the active configuration file, 3) starting the program again, and 4) manually "loading" the renamed configuration file. I was then able to re-select the sensors like before.

DKogut17, 2024-03-12:
I'm also experiencing this issue with using any sensors in HWinfo. It works for a moment but the second it gets a NaN it fails and records an error in the log.

lgraak, 2024-03-12:
> I'm also experiencing this issue with using any sensors in HWinfo. It works for a moment but the second it gets a NaN it fails and records an error in the log.

Try hiding sensors you don't need in hwinfo. I think there is an issue somewhere with how many sensors are being brought in from HWinfo. After I pruned the sensors, this issue went away.

DKogut17, 2024-03-12:
> Try hiding sensors you don't need in hwinfo. I think there is an issue somewhere with how many sensors are being brought in from HWinfo. After I pruned the sensors, this issue went away.

So what ended up working so far for me is to fully reinstall HWinfo because I noticed in the registry that the entire key/value pair for the MemTemp sensor was disappearing when I was refreshing constantly. So the NaN was a true reading it was just what HWinfo was reading at the time. So far it's been working since that.