Rem0o/FanControl.HWInfo #41: Value not parsed

Reveller500, 2023-04-23:
Hi, 

I get this error in the FanControl log when trying to fetch temperatures from HWinfo: "Value not parsed: 33,5". Values from HWinfo show as "NaN" in FanControl. If I change from comma decimal separator to dot in HWinfo setting the error disappears. But I wish to use comma decimal. Anything that can be solved in future update?
I'm on version 154 and HWinfo64 v7.42-5030 with "shared memory" disabled and gadget active. 

Thanks for a fantastic software!
Kind regards

Rem0o, 2023-04-23:
See

https://github.com/Rem0o/FanControl.HWInfo/blob/077db8f0c09c9ed92ab5c976c5623e2a753157d4/HWInfoRegistry.cs#L11

Reveller500, 2023-04-23:
Thank you for the speedy reply. Could you please offer some guidance on how I can use this info?

closed 2026-09-03 by Rem0o