Rem0o/FanControl.HWInfo #45: System.Exception: HWInfo sensors failed

FlemmingJPetersen, 2023-06-22:
2023-06-22 15.21.12: HWInfo sensor failed momentarily during operation: HWInfo/ASUS TUF GAMING Z690-PLUS D4 (Nuvoton NCT6798D)/CPU/RPM - Missing
2023-06-22 15.21.12: Unhandled exception in FanControl v160.0.0.0
2023-06-22 15.21.12: System.Exception: HWInfo sensors failed: HWInfo/ASUS TUF GAMING Z690-PLUS D4 (Nuvoton NCT6798D)/CPU/RPM
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 93
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

mminehanNZ, 2023-07-29:
I'm seeing the same error:

> 9/07/2023 8:43:25 am: HWInfo sensor failed momentarily during operation: HWInfo/ASUS EC: ASUS /CPU Core Current/A - Missing
> 29/07/2023 8:43:25 am: Unhandled exception in FanControl v164.0.0.0
> 29/07/2023 8:43:25 am: System.Exception: HWInfo sensors failed: HWInfo/ASUS EC: ASUS /CPU Core Current/A
>    at FanControl.HWInfo.HWInfoPlugin.Update()
>    at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
>    at System.Collections.Generic.List`1.ForEach(Action`1 action)
>    at FanControl.Domain.ComputerAccessLayer.Update()
>    at FanControl.Domain.ApplicationClock.DoActions()
> --- End of stack trace from previous location where exception was thrown ---
>    at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
>    at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
>    at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
> 29/07/2023 11:30:03 am: HWInfo sensor failed momentarily during operation: HWInfo/ASUS EC: ASUS /CPU Core Current/A - Missing
> 29/07/2023 11:30:03 am: Unhandled exception in FanControl v164.0.0.0
> 29/07/2023 11:30:03 am: System.Exception: HWInfo sensors failed: HWInfo/ASUS EC: ASUS /CPU Core Current/A
>    at FanControl.HWInfo.HWInfoPlugin.Update()
>    at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
>    at System.Collections.Generic.List`1.ForEach(Action`1 action)
>    at FanControl.Domain.ComputerAccessLayer.Update()
>    at FanControl.Domain.ApplicationClock.DoActions()
> --- End of stack trace from previous location where exception was thrown ---
>    at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
>    at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
>    at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

mminehanNZ, 2023-08-11:
Any update or progress on this issue?
I'm getting this error almost hourly.

mminehanNZ, 2023-08-16:
So I have kind of fixed the problem by disabling the CPU current draw measurement in the HWInfo app itself:
29/07/2023 11:30:03 am: HWInfo sensor failed momentarily during operation: HWInfo/ASUS EC: ASUS /CPU Core Current/A - Missing

Seems the fan control plugin has issues if one of the HWInfo sensors goes missing.

gund, 2024-07-10:
This is also happening on my machine, I'm monitoring HWInfo sensors of coolant, ram and drives temperatures but they periodically fail and read zeros.
The missing sensor seems to be one of the RAM sticks temperature `HWInfo/DDR5 DIMM [#1] (BANK 0/Controller0-DIMM1)/SPD Hub Temperature/°C` and not the CPU current draw as in other case so this seems to be a random failure.

Also since the HWInfo itself has no issues with this sensor and always reads the values without dropping to zeros maybe Fancontrol plugin can gracefully handle this error instead of a hard failure and retry the sensor in a few milliseconds?
This retry timeout could even be configurable in the UI if it's also non deterministic and different machines would need different timeouts.

Here are the logs:
```
10/07/2024 01:28:58: HWInfo sensor failed momentarily during operation: HWInfo/DDR5 DIMM [#1] (BANK 0/Controller0-DIMM1)/SPD Hub Temperature/°C - Missing
HWInfo/DDR5 DIMM [#3] (BANK 0/Controller1-DIMM1)/SPD Hub Temperature/°C - Missing
10/07/2024 01:28:58: Unhandled exception in FanControl v194.0.0.0
10/07/2024 01:28:58: LogUnhandledException - System.Exception: HWInfo sensors failed: HWInfo/DDR5 DIMM [#1] (BANK 0/Controller0-DIMM1)/SPD Hub Temperature/°C
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 93
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
10/07/2024 02:18:02: HWInfo sensor failed momentarily during operation: HWInfo/DDR5 DIMM [#1] (BANK 0/Controller0-DIMM1)/SPD Hub Temperature/°C - Missing
10/07/2024 02:18:02: Unhandled exception in FanControl v194.0.0.0
10/07/2024 02:18:02: LogUnhandledException - System.Exception: HWInfo sensors failed: HWInfo/DDR5 DIMM [#1] (BANK 0/Controller0-DIMM1)/SPD Hub Temperature/°C
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 93
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
10/07/2024 02:45:24: HWInfo sensor failed momentarily during operation: HWInfo/DDR5 DIMM [#1] (BANK 0/Controller0-DIMM1)/SPD Hub Temperature/°C - Missing
HWInfo/DDR5 DIMM [#3] (BANK 0/Controller1-DIMM1)/SPD Hub Temperature/°C - Missing
10/07/2024 02:45:24: Unhandled exception in FanControl v194.0.0.0
10/07/2024 02:45:24: LogUnhandledException - System.Exception: HWInfo sensors failed: HWInfo/DDR5 DIMM [#3] (BANK 0/Controller1-DIMM1)/SPD Hub Temperature/°C, HWInfo/DDR5 DIMM [#1] (BANK 0/Controller0-DIMM1)/SPD Hub Temperature/°C
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 93
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
10/07/2024 15:32:02: HWInfo sensor failed momentarily during operation: HWInfo/DDR5 DIMM [#1] (BANK 0/Controller0-DIMM1)/SPD Hub Temperature/°C - Missing
HWInfo/DDR5 DIMM [#3] (BANK 0/Controller1-DIMM1)/SPD Hub Temperature/°C - Missing
10/07/2024 15:32:02: Unhandled exception in FanControl v194.0.0.0
10/07/2024 15:32:02: LogUnhandledException - System.Exception: HWInfo sensors failed: HWInfo/DDR5 DIMM [#1] (BANK 0/Controller0-DIMM1)/SPD Hub Temperature/°C, HWInfo/DDR5 DIMM [#3] (BANK 0/Controller1-DIMM1)/SPD Hub Temperature/°C
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 93
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at System.Collections.Generic.List`1.ForEach(Action`1 action)
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
[... cut ...]
   at FanControl.Domain.ComputerAccessLayer.Update()
   at FanControl.Domain.ApplicationClock.DoActions()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
```