Rem0o/FanControl.HWInfo #46: Plugin not working failed to initialize

Rexadev, 2023-08-21:
[image]

Rexadev, 2023-08-25:
@Rem0o Please reopen this

Rem0o, 2023-08-25:
In the case of HWINFO, have you gone through the usual suspects?

Is HWInfo running, do you have temperatures sensors exported (Report value in Gadget) from there?

Rexadev, 2023-08-25:
> Is HWInfo running, 

Yes the portable version

> do you have temperatures sensors exported (Report value in Gadget) from there?

How ?

@Rem0o

Rem0o, 2023-08-25:
From the README:

```
Go the HWInfo's "Configure Sensors" option menu, then to the "HWInfo Gadget" tab. Have the "Enable reporting to gadget" checkbox checked and select "Report value in Gadget" for all sensors you want to import.
```

Rexadev, 2023-08-25:
### After installing HWInfo
[image]

### Are there more than 1 HWInfo
[image]

Rem0o, 2023-08-25:
It's in the "sensor" section.

Why do you want to use this plugin if you never used HWInfo before?

Rexadev, 2023-08-25:
add it to readme @Rem0o

Rexadev, 2023-08-25:
After installing HWInfo [image]
@Rem0o

Rem0o, 2023-08-25:
@Rexadev  this repo shouldn't be a tutorial on how to use HWInfo. There are plenty of resource around for that.

Rexadev, 2023-08-26:
> After installing HWInfo [image] @Rem0o

@Rem0o this related to FanControl