Rem0o/FanControl.HWInfo #48: Can't get the plugin to work

ArolaM, 2023-09-02:
2.9.2023 16.40.21: HWInfo could not initialize or has no sensors.
2.9.2023 16.47.20: Unhandled exception in FanControl v168.0.0.0
2.9.2023 16.47.20: System.Exception: HWInfo was closed during operation.
   kohteessa FanControl.HWInfo.HWInfoPlugin.Update()
   kohteessa FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   kohteessa System.Collections.Generic.List`1.ForEach(Action`1 action)
   kohteessa FanControl.Domain.ComputerAccessLayer.Update()
   kohteessa FanControl.Domain.ApplicationClock.DoActions()
--- Pinon jäljityksen loppu aiemmasta sijainnista, jossa palautettiin poikkeus ---
   kohteessa System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   kohteessa System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   kohteessa System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
2.9.2023 16.49.41: Unhandled exception in FanControl v168.0.0.0
2.9.2023 16.49.41: System.Exception: HWInfo was closed during operation.
   kohteessa FanControl.HWInfo.HWInfoPlugin.Update()
   kohteessa FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   kohteessa System.Collections.Generic.List`1.ForEach(Action`1 action)
   kohteessa FanControl.Domain.ComputerAccessLayer.Update()
   kohteessa FanControl.Domain.ApplicationClock.DoActions()
--- Pinon jäljityksen loppu aiemmasta sijainnista, jossa palautettiin poikkeus ---
   kohteessa System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   kohteessa System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   kohteessa System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

"HWInfo could not initialize or has no sensors."

Rem0o, 2023-09-04:
Is HWInfo running? Did you export sensors properly?

ArolaM, 2023-09-04:
> Is HWInfo running? Did you export sensors properly?

I wouldn't be asking this if these steps weren't taken ;)

Rem0o, 2023-09-05:
Do you get the proper registry keys appearing ?

See https://github.com/Rem0o/FanControl.HWInfo/blob/master/HWInfoRegistry.cs

ArolaM, 2023-09-05:
> Do you get the proper registry keys appearing ?
> 
> See https://github.com/Rem0o/FanControl.HWInfo/blob/master/HWInfoRegistry.cs

Yes, location "*\HKEY_CURRENT_USER\SOFTWARE\HWiNFO64\VSB" has registry keys there.

See attached image for reference.

[image]

However, my system language is in Finnish. Could this be a reason?

ArolaM, 2023-09-05:
Okay, I discovered that my initial plan of using GPU and CPU usage percentages will not work. When I tried using only GPU and CPU package temps, they came visible.

Any way to play with the usage percentage?

ArolaM, 2023-09-05:
> Okay, I discovered that my initial plan of using GPU and CPU usage percentages will not work. When I tried using only GPU and CPU package temps, they came visible.
> 
> Any way to play with the usage percentage?

Okay, with a little tinkering by changing the "%" sign to "°C" makes it possible.
[image]

Acelya-9028, 2024-06-02:
Nice, thank you very much for the solution :D