Rem0o/FanControl.HWInfo #54: No Binaries on AppVeyor

ApBrown1298, 2024-01-28:
I was trying to get this running but when I go to the AppVeyor link, there's nothing under "Artifacts". Am I missing something?

Fr0stX76, 2024-01-28:
Try [here](https://github.com/Rem0o/FanControl.HWInfo/releases/download/1.3/Release.zip)

ApBrown1298, 2024-01-30:
Ok I may be  lost on a different part then... if the binaries are included in the Release.zip then what am I supposed to do after that? I opened the FanControl.HWInfo.csproj file in Visual Studio, built the solution, and then copied the .dll over but FanControl still shows that there are no plugins.

Any thoughts on where I went wrong?

ApBrown1298, 2024-01-30:
Thank you both for the help... on top of not having it unlocked I was a bit of a smooth brain and had two instances of FanControl on different drives. 

Works perfectly now, thanks again!