Rem0o/FanControl.HWInfo #59: HWinfo plugin sensor shows in settings but no fans other than GPU are detected in Fancontrol

Kijicode, 2024-09-22:
Could someone please ELI5 how to get Fancontrol show my Dell desktop PC's fans in fanctrol? 

- HWinfo is running and connected under plugins (sensor info is showing, see screenshots). 
- But no matter how often I run the setup assistant in Fancontrol, no fans other than the GPU are appearing. 

Some other post claimed this plugin might not work with newer versions of fancontrol, but I could not find a download anymore for version 147 that supposedly works.

I'm extremely unfamiliar with setups like this and a noob, so it's already taken me literally hours to get here...
Thanks in advance!

HWiNFO64 v8 10-5520 - Sensors Status shows 2 tower fans
[image]

HWiNFO64 - Sensor Settings - the 2 tower fans should be the docking ones
[image]

Fan Control V203 plugin sensors
[image]

Fan Control V203 no other fans than GPU recognised
[image]

Fr0stX76, 2024-09-22:
It work as intended. The HWinfoplugin is meant to detect temp sensors only.

If fancontrol doesn't detect your fans, it's either because Librehardwaremonitor doesn't support them or because you didn't enable something needed under setting/sensors/edit sources

My guess is that Librehardwaremonitor doesn't support them. An easy way to prove it, is to install the latest build of Librehardwaremonitor [here](https://nightly.link/LibreHardwareMonitor/LibreHardwareMonitor/workflows/master/master/LibreHardwareMonitor-net472.zip) and see if they get detected.

Finally, you could try the [DellPlugin](https://github.com/Rem0o/FanControl.DellPlugin). But don't put too much hope on this one...

Kijicode, 2024-09-22:
> It work as intended. The HWinfoplugin is meant to work with temp sensors only.
> 
> If fancontrol doesn't detect your fans, it's either because Librehardwaremonitor doesn't support them or because you didn't enable something needed under setting/sensors/edit sources
> 
> My guess is that Librehardwaremonitor doesn't support them. An easy way to prove it, is to install the latest build of Librehardwaremonitor [here](https://nightly.link/LibreHardwareMonitor/LibreHardwareMonitor/workflows/master/master/LibreHardwareMonitor-net472.zip) and see if they get detected.

Thanks, I didn't know that. LibreHardwareMonitor indeed only shows the GPU fan

> Finally, you could try the [DellPlugin](https://github.com/Rem0o/FanControl.DellPlugin). But don't put too much hope on this one...
Sadly that doesn't seem to work either. With the dell plugin fancontrol fails to initialize, but perhaps it will get better over time.

Anyway, thanks for your answer! Now I can give up on trying to make fancontrol work for my crappy Dell G5.