Rem0o/FanControl.HWInfo #63: Missing sensors (watts, volts, percentages)

FlopRe, 2024-11-30:
My desired configuration is to have CPU, GPU and Memory Total Power (W) all report to FanControl, so that I can combine these using customs sensors for a very rough total system power usage, then display these as tray icons.

After attempting to configure this, I noticed temperature sensors are being picked up by the HWInfo plugin, but not any other values such as watts, volts, percentages or file size units (KB, MB, GB).

I'm getting no errors in the logs, and I've confirmed HWInfo is correctly reporting these values into registry:
[image]

I believe this may just be how the plugin was programmed so I ask, could you please add support for these values, so these can be combined in custom sensors, and shown as tray icons?

Rem0o, 2024-12-01:
FanControl itself ( the main program ) doesn't support other types of sensors except temperatures as inputs for fan curves, tray icons, custom sensors and so on.  The HWInfo plugin thus only export temperature sensors ( C or F ).

Adding power sensors would introduce a lot of complications. Display, value ranges, combining both types in custom sensors, hysteresis and response time would have to be reworked and I could go on and on. 

If you want to hack your way into doing it and trying it, you can change the unit type for a given sensor in HWInfo, so export a changed sensor from W to C, and FanControl will pick it up. Do some tests with that and see how that goes.

FlopRe, 2024-12-02:
That worked a treat, thank you! 
[image]

Would be great to have this functionality in base FanControl for the ultimate HWInfo replacement, but I understand how much work it would be. Either way, thank you for this godsent of a program 🔥

closed 2024-12-02 by FlopRe