Rem0o/FanControl.HWInfo #67: Sensors are detected and when refreshing in FanControl, detection works for about 2 seconds, then indefinitely displays "-°C" instead of detected temperature.

ParanormalBanana, 2025-03-12:
basically the title. Tried to upload recording of whzt happens but can't

Rem0o, 2025-03-12:
See https://github.com/Rem0o/FanControl.Releases/issues/3016

ParanormalBanana, 2025-03-12:
Hi, I don't see how this is the same issue at all? I have no issues whatsoever with Windows Defender or FanControl in general, this only happens with FanControl.HWInfo... I understand nothing about the issue described in the reffered post.

Rem0o, 2025-03-12:
Sorry I got absolutely bombarded with similar issue since the driver in FanControl started being blocked so I was going through similar issues swiftly.

So the sensors you are talking about are the exported ones from HWInfo ? Mind sharing a screenshot?

ApBrown1298, 2025-03-16:
Not the OP but I have the same issue. I have an Intel A750, the sensors of which I'm trying to export from HWInfo. I attached a few screenshots, a couple from HWInfo and one from Fan Control. I also took a video of the refresh behavior.

HWInfo Sensor Readings
[image]

HWInfo Sensor Settings
[image]

Fan Control Detecting Sensors from HWInfo
[image]

Fan Control Refresh Behavior
https://github.com/user-attachments/assets/552acc8c-be16-4f51-a68e-0206684db38d

Fan Control Log File
[log.txt](https://github.com/user-attachments/files/19268728/log.txt)

ParanormalBanana, 2025-03-16:
Hi, I've found a workaround to the issue after tinkering for a bit. This issue happened only when I had the "Thermal Throttling" metric enabled in HWInfo Gadget. When I disabled it, it began running without an issue. 
I only tried to use Total CPU Usage, Max CPU usage, GPU Utilization and Thermal Throttling, so I don't know if there are other metrics that cause the issue, or why it did. I originally wanted to use Thermal Throttling metrics in order to be able to run my fans really low, but have a backup fan speed boost if throttling, as a means to inform me that it was throttling in the tweaking phase. It seems that for some reason, FanControl doesn't like these metrics and shuts down the entire HWInfo Gadget if it's enabled. 

For Now, I simply have it disabled, but I was thinking of building a workaround involving monitoring the CPU frequency as a replacement for the throttling indicator, but haven't yet got around to doing it, so I don't know if that would work. 

When I had the issue, at some point I found the log from FanControl and was able to open it and look at it, but I forgot to save it and don't remember what it said,, although I can say it definitely said something about the throttling metric.

ParanormalBanana, 2025-03-16:
Edit: Nevermind, I had saved the log file, so here's what it said:

3/12/2025 7:39:16 AM: HWInfo sensor failed momentarily during operation: HWInfo/CPU [#0]: AMD Ryzen 9 5900X: Enhanced/Thermal Throttling (HTC)/ - Value not parsed: No
3/12/2025 7:39:16 AM: Update - System.Exception: HWInfo sensors failed: HWInfo/CPU [#0]: AMD Ryzen 9 5900X: Enhanced/Thermal Throttling (HTC)/
   at FanControl.HWInfo.HWInfoPlugin.Update() in C:\projects\fancontrol-hwinfo\HWInfoPlugin.cs:line 93
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Update()
   at FanControl.Domain.ComputerAccessLayer.Update()

ApBrown1298, 2025-03-16:
Ah ok, I looked through my log file as well and it seemed to not like the "Thermal" metric. There's two of those attached to the A750 in HWInfo that both had reporting enabled, so I turned them off and now it's working. Thanks for the tip!

I think I might have turned them on to see if FanControl would bring them in because I have no clue what the "Thermal" metric does, but it didn't bring them in so I didn't think anything of it.

dapperstache, 2025-06-04:
Yeah, anything that is "yes" or "no" seems to have issues. I was able to get it working, but it doesn't like my Octo very much.

PyLit, 2025-08-27:
> Yeah, anything that is "yes" or "no" seems to have issues. I was able to get it working, but it doesn't like my Octo very much if you fiddle with it's settings in Fan Control. OK to set and forget.
> 
> Kind of sucks, because would like to output Drive Failure and warning alerts to Rainmeter from HWiNFO, but with current configuration, that just isn't possible.

This fixed it for me. Seems to only accept temp and I guess fan RPM. I accidentally had Chassis Intrusion set to report in gadget which apparently breaks all the other sensors on the FanControl Side. Also too bad about the rainmeter conflict

scandinavia1500-jpn, 2026-11-03:
Thank you, ParanormalBanana, for identifying this issue and sharing your solution!

We encountered exactly the same problem on an ASUS RAMPAGE IV FORMULA motherboard with an Intel Xeon E5-2697 v2 and an NVIDIA GTX 1070.

Our configuration:
- Windows 11
- HWiNFO64 v8.50-6020
- Fan Control v275
- FanControl.HWInfo plugin

The temperature sensor (OPT1 via ASUS EC) appeared correctly for about two seconds after launching Fan Control, then disappeared.

Following your findings, we discovered that four NVIDIA GPU Performance Limit sensors were causing the problem:

- GPU Performance Limit
- Performance Limit – Power
- Performance Limit – Thermal
- Performance Limit – Utilization

We tested each of these four sensors individually. Enabling Gadget reporting for any one of them caused the OPT1 temperature reading to disappear.

Disabling Gadget reporting for all four resolved the issue, while GPU temperature and fan RPM monitoring remained available.

Your investigation helped us solve a very frustrating problem in our X79 hardware research project in Japan.

Thank you again for sharing your findings with the community!

ParanormalBanana, 2026-11-03:
Glad my tinkering was able to help someone else :)