Rem0o/FanControl.HWInfo #68: Since HWiNFO 8.26 update T_Sen1 is not reported back to Fac Control

Der-Orso, 2025-05-19:
Since HWiNFO 8.26 update T_Sen1 is not reported back to Fac Control.

HWiNFO 8.26 starts without issue. FC has a 10 second delay to allow for HWi to be certain to be started. FC pops open (set to start in system tray), pop-up says,, "(!) Error.. ..missing sensor with identifier: HWiNFO/HWiNFO/MSI MEG X670E ACE (MS-7D69) (Nuvoton NCT6687D)/T_SEN1/^C)

In system tray, I right-click => HWiNFO  => Sensors (HWiNFO opens on the Sensors page) => Save Settings and Quit

HWiNFO  saves and closes.

Pop-up in FC: click [Refresh],  FC goes thru its refreshing and at the end the T_Sen1 is found again and everything works.

Windows 10 64bit
Fan Control 224
FC/HWiNFO latest
HWiNFO 8.26 64bit installed

Rem0o, 2025-05-19:
FanControl only scans the sensors on start. My guess is that this T_SEN1 sensor doesn't show up in HWInfo initially?

Der-Orso, 2025-05-23:
Or something in HWiNFO has changed the way it works with the FC.HWiNFO gadget.

If, before FC has started--10 seconds delay--, I open the HWiNFO Sensors page and immediately click on Save & Quit, when FC starts, the T_Sensor is found and everything works.

As I stated before, this only started after updating HWiNFO to 8.26 from 8.24.